geerlingguy/ansible-role-docker_arm #7: Seems the newest version of docker-compose can not be installed via pip

natelandau, 2020-01-26:
Not an issue with this repo per se, but the playbook fails to install docker-compose via pip.  See this thread: https://github.com/docker/compose/issues/6617

```
$ pip install docker-compose                                                                                                                                                                     
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting docker-compose
  Downloading https://files.pythonhosted.org/packages/f8/fa/94a68c7fb1292a726e95be53c8a66a85370de74b5d68adf7fcef61e6c7c9/docker_compose-1.25.3-py2.py3-none-any.whl (136kB)
    100% |################################| 143kB 1.8MB/s
Requirement already satisfied: backports.ssl-match-hostname<4,>=3.5; python_version < "3.5" in /usr/local/lib/python2.7/dist-packages (from docker-compose) (3.7.0.1)
Requirement already satisfied: six<2,>=1.3.0 in /usr/lib/python2.7/dist-packages (from docker-compose) (1.12.0)
Collecting docopt<1,>=0.6.1 (from docker-compose)
  Downloading https://www.piwheels.org/simple/docopt/docopt-0.6.2-py2.py3-none-any.whl
Collecting dockerpty<1,>=0.4.1 (from docker-compose)
  Downloading https://files.pythonhosted.org/packages/8d/ee/e9ecce4c32204a6738e0a5d5883d3413794d7498fe8b06f44becc028d3ba/dockerpty-0.4.1.tar.gz
Requirement already satisfied: docker[ssh]<5,>=3.7.0 in /usr/local/lib/python2.7/dist-packages (from docker-compose) (4.1.0)
Requirement already satisfied: requests<3,>=2.20.0 in /usr/local/lib/python2.7/dist-packages (from docker-compose) (2.22.0)
Collecting texttable<2,>=0.9.0 (from docker-compose)
  Downloading https://files.pythonhosted.org/packages/ec/b1/8a1c659ce288bf771d5b1c7cae318ada466f73bd0e16df8d86f27a2a3ee7/texttable-1.6.2-py2.py3-none-any.whl
Collecting subprocess32<4,>=3.5.4; python_version < "3.2" (from docker-compose)
  Downloading https://files.pythonhosted.org/packages/32/c8/564be4d12629b912ea431f1a50eb8b3b9d00f1a0b1ceff17f266be190007/subprocess32-3.5.4.tar.gz (97kB)
    100% |################################| 102kB 2.9MB/s
Collecting cached-property<2,>=1.2.0 (from docker-compose)
  Downloading https://files.pythonhosted.org/packages/3b/86/85c1be2e8db9e13ef9a350aecd6dea292bd612fa288c2f40d035bb750ded/cached_property-1.5.1-py2.py3-none-any.whl
Collecting PyYAML<6,>=3.10 (from docker-compose)
  Downloading https://files.pythonhosted.org/packages/3d/d9/ea9816aea31beeadccd03f1f8b625ecf8f645bd66744484d162d84803ce5/PyYAML-5.3.tar.gz (268kB)
    100% |################################| 276kB 1.3MB/s
Requirement already satisfied: websocket-client<1,>=0.32.0 in /usr/local/lib/python2.7/dist-packages (from docker-compose) (0.57.0)
Requirement already satisfied: enum34<2,>=1.0.4; python_version < "3.4" in /usr/lib/python2.7/dist-packages (from docker-compose) (1.1.6)
Requirement already satisfied: ipaddress<2,>=1.0.16; python_version < "3.3" in /usr/lib/python2.7/dist-packages (from docker-compose) (1.0.17)
Collecting jsonschema<4,>=2.5.1 (from docker-compose)
  Downloading https://files.pythonhosted.org/packages/c5/8f/51e89ce52a085483359217bc72cdbf6e75ee595d5b1d4b5ade40c7e018b8/jsonschema-3.2.0-py2.py3-none-any.whl (56kB)
    100% |################################| 61kB 2.4MB/s
Collecting backports.shutil-get-terminal-size==1.0.0; python_version < "3.3" (from docker-compose)
  Downloading https://files.pythonhosted.org/packages/7d/cd/1750d6c35fe86d35f8562091737907f234b78fdffab42b29c72b1dd861f4/backports.shutil_get_terminal_size-1.0.0-py2.py3-none-any.whl
Collecting paramiko>=2.4.2; extra == "ssh" (from docker[ssh]<5,>=3.7.0->docker-compose)
  Downloading https://files.pythonhosted.org/packages/06/1e/1e08baaaf6c3d3df1459fd85f0e7d2d6aa916f33958f151ee1ecc9800971/paramiko-2.7.1-py2.py3-none-any.whl (206kB)
    100% |################################| 215kB 1.7MB/s
Requirement already satisfied: idna<2.9,>=2.5 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.20.0->docker-compose) (2.8)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.20.0->docker-compose) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.20.0->docker-compose) (2019.11.28)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.20.0->docker-compose) (1.25.8)
Requirement already satisfied: setuptools in /usr/lib/python2.7/dist-packages (from jsonschema<4,>=2.5.1->docker-compose) (40.8.0)
Collecting attrs>=17.4.0 (from jsonschema<4,>=2.5.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/a2/db/4313ab3be961f7a763066401fb77f7748373b6094076ae2bda2806988af6/attrs-19.3.0-py2.py3-none-any.whl
Collecting importlib-metadata; python_version < "3.8" (from jsonschema<4,>=2.5.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/d7/31/74dcb59a601b95fce3b0334e8fc9db758f78e43075f22aeb3677dfb19f4c/importlib_metadata-1.4.0-py2.py3-none-any.whl
Collecting pyrsistent>=0.14.0 (from jsonschema<4,>=2.5.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/90/aa/cdcf7ef88cc0f831b6f14c8c57318824c9de9913fe8de38e46a98c069a35/pyrsistent-0.15.7.tar.gz (107kB)
    100% |################################| 112kB 2.9MB/s
Collecting functools32; python_version < "3" (from jsonschema<4,>=2.5.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/c5/60/6ac26ad05857c601308d8fb9e87fa36d0ebf889423f47c3502ef034365db/functools32-3.2.3-2.tar.gz
Collecting bcrypt>=3.1.3 (from paramiko>=2.4.2; extra == "ssh"->docker[ssh]<5,>=3.7.0->docker-compose)
  Downloading https://files.pythonhosted.org/packages/fa/aa/025a3ab62469b5167bc397837c9ffc486c42a97ef12ceaa6699d8f5a5416/bcrypt-3.1.7.tar.gz (42kB)
    100% |################################| 51kB 3.5MB/s
  Installing build dependencies ... error
  Complete output from command /usr/bin/python -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-kGJqHF --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools>=40.8.0 wheel "cffi>=1.1; python_implementation != 'PyPy'":
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=40.8.0
    Downloading https://files.pythonhosted.org/packages/f9/d3/955738b20d3832dfa3cd3d9b07e29a8162edb480bf988332f5e6e48ca444/setuptools-44.0.0-py2.py3-none-any.whl (583kB)
  Collecting wheel
    Downloading https://files.pythonhosted.org/packages/00/83/b4a77d044e78ad1a45610eb88f745be2fd2c6d658f9798a15e384b7d57c9/wheel-0.33.6-py2.py3-none-any.whl
  Collecting cffi>=1.1
    Downloading https://files.pythonhosted.org/packages/2d/bf/960e5a422db3ac1a5e612cb35ca436c3fc985ed4b7ed13a1b4879006f450/cffi-1.13.2.tar.gz (460kB)
  Collecting pycparser (from cffi>=1.1)
    Downloading https://www.piwheels.org/simple/pycparser/pycparser-2.19-py2.py3-none-any.whl (111kB)
  Building wheels for collected packages: cffi
    Running setup.py bdist_wheel for cffi: started
    Running setup.py bdist_wheel for cffi: finished with status 'error'
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-nZ1y7L/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-i_cv6F --python-tag cp27:
    Package libffi was not found in the pkg-config search path.
[... 104 lines cut ...]
      copying cffi/error.py -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/cffi_opcode.py -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/_cffi_include.h -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/parse_c_type.h -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/_embedding.h -> build/lib.linux-armv7l-2.7/cffi
      copying cffi/_cffi_errors.h -> build/lib.linux-armv7l-2.7/cffi
      running build_ext
      building '_cffi_backend' extension
      creating build/temp.linux-armv7l-2.7
      creating build/temp.linux-armv7l-2.7/c
      arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-InigCj/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/u [... cut ...] " failed with error code 1 in /tmp/pip-install-nZ1y7L/cffi/

  ----------------------------------------
Command "/usr/bin/python -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-kGJqHF --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools>=40.8.0 wheel "cffi>=1.1; python_implementation != 'PyPy'"" failed with error code 1 in None
```

I believe you need to pin the version

danmilon, 2020-02-26:
We should be able to use the pre-built wheels on pypi.

geerlingguy, 2020-03-25:
Is this only the case on Raspbian 10? Or is this happening in regular old Debian as well?

The Debian 10 tests seem to pass without issue: https://travis-ci.com/github/geerlingguy/ansible-role-docker_arm/jobs/299281915

danmilon, 2020-03-26:
I got this on armbian buster.

mikev1963, 2020-03-28:
I installed the dependency as described above, but now I get the following error:
_$ docker-compose --version_
```
Traceback (most recent call last):
  File "/usr/local/bin/docker-compose", line 6, in <module>
    from compose.cli.main import main
  File "/usr/local/lib/python2.7/dist-packages/compose/cli/main.py", line 18, in <module>
    import docker.errors
  File "/usr/local/lib/python2.7/dist-packages/docker/__init__.py", line 2, in <module>
    from .api import APIClient
  File "/usr/local/lib/python2.7/dist-packages/docker/api/__init__.py", line 2, in <module>
    from .client import APIClient
  File "/usr/local/lib/python2.7/dist-packages/docker/api/client.py", line 10, in <module>
    from .build import BuildApiMixin
  File "/usr/local/lib/python2.7/dist-packages/docker/api/build.py", line 6, in <module>
    from .. import auth
  File "/usr/local/lib/python2.7/dist-packages/docker/auth.py", line 9, in <module>
    from .utils import config
  File "/usr/local/lib/python2.7/dist-packages/docker/utils/__init__.py", line 3, in <module>
    from .decorators import check_resource, minimum_version, update_headers
  File "/usr/local/lib/python2.7/dist-packages/docker/utils/decorators.py", line 4, in <module>
    from . import utils
  File "/usr/local/lib/python2.7/dist-packages/docker/utils/utils.py", line 13, in <module>
    from .. import tls
  File "/usr/local/lib/python2.7/dist-packages/docker/tls.py", line 5, in <module>
    from .transport import SSLHTTPAdapter
  File "/usr/local/lib/python2.7/dist-packages/docker/transport/__init__.py", line 3, in <module>
    from .ssladapter import SSLHTTPAdapter
  File "/usr/local/lib/python2.7/dist-packages/docker/transport/ssladapter.py", line 23, in <module>
    from backports.ssl_match_hostname import match_hostname
ImportError: No module named ssl_match_hostname

```

whoan, 2020-03-30:
@mikev1963 You need to use the distribution's package *backports.ssl-match-hostname* instead of the one from pip. See #4 for more information.

commit 2b7bae6b (2020-04-01): Fixes #7, PR #4: Fix Python 2-related docker-compose installation issues.
closed 2020-04-01 by geerlingguy