geerlingguy/ansible-role-docker_arm #8: Primary group for user is changed to docker

ejsmit, 2020-04-28:
The current role currently uses the following to change the group membership:
```
  user:
    name: "{{ item }}"
    group: docker
    append: true
  with_items: "{{ docker_users }}"
```

which cause this to happen:
```
$ id
uid=1000(pi) gid=995(docker) groups=995(docker),4(adm),  ...
```

The `group`  should be `groups`.  This will leave the user in its original primary group (pi, gid 1000) and only add the user to the docker group.  

I also checked your docker role.  It correctly uses groups.

geerlingguy, 2020-04-29:
You're correct, thanks for catching this oversight! I'll have this fixed ASAP.

geerlingguy, 2020-04-29:
This is available in 3.1.0.

commit b38ae898 (2020-04-29): Fixes #8: Use groups instead of group to preserve primary group for docker users.
closed 2020-04-29 by geerlingguy