geerlingguy/ansible-role-docker_arm #11: Role fails on Raspberry Pi OS May 2020 image

jtcmedia, 2020-05-30:
Running this role in Ansible produces:
```
fatal: [k3s-master]: FAILED! => {"cache_update_time": 1590854762, "cache_updated": false, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"   --force-yes   install 'docker-ce=5:19.03.8~3-0~raspbian-buster'' failed: W: --force-yes is deprecated, use one of the options starting with --allow instead.\nE: Sub-process /usr/bin/dpkg returned an error code (1)\n", "rc": 100, "stderr": "W: --force-yes is deprecated, use one of the options starting with --allow instead.\nE: Sub-process /usr/bin/dpkg returned an error code (1)\n", "stderr_lines": ["W: --force-yes is deprecated, use one of the options starting with --allow instead.", "E: Sub-process /usr/bin/dpkg returned an error code (1)"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following additional packages will be installed:\n  aufs-dkms aufs-tools dkms\nSuggested packages:\n  aufs-dev python3-apport menu\nRecommended packages:\n  cgroupfs-mount | cgroup-lite libltdl7\nThe following NEW packages will be installed:\n  aufs-dkms aufs-tools dkms\nThe following packages will be DOWNGRADED:\n  docker-ce\n0 upgraded, 3 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.\nNeed to get 0 B/15.3 MB of archives.\nAfter this operation, 5305 kB of additional disk space will be used.\nSelecting previously unselected package dkms.\r\n(Reading database ... \r(Reading database ... 5%\r(Reading database ... 10%\r(Reading database ... 15%\r(Reading database ... 20%\r(Reading database ... 25%\r(Reading database ... 30%\r(Reading database ... 35%\r(Reading database ... 40%\r(Reading database ... 45%\r(Reading database ... 50%\r(Reading database ... 55%\r(Reading database ... 60%\r(Reading database ... 65%\r(Reading database ... 70%\r(Reading database ... 75%\r(Reading database ... 80%\r(Reading database ... 85%\r(Reading database ... 90%\r(Reading database ... 95%\r(Reading database ... 100%\r(Reading database ... 97112 files and directories currently installed.)\r\nPreparing to unpack .../archives/dkms_2.6.1-4_all.deb ...\r\nUnpacking dkms (2.6.1-4) ...\r\nSelecting previously unselected package aufs-dkms.\r\nPreparing to unpack .../aufs-dkms_4.19+20190211-1_all.deb ...\r\nUnpacking aufs-dkms (4.19+20190211-1) ...\r\nSelecting previously unselected package aufs-tools.\r\nPreparing to unpack .../aufs-tools_1%3a4.14+20190211-1_armhf.deb ...\r\nUnpacking aufs-tools (1:4.14+20190211-1) ...\r\ndpkg: warning: downgrading docker-ce from 5:19.03.10~3-0~raspbian-buster to 5:19.03.8~3-0~raspbian-buster\r\nPreparing to unpack .../docker-ce_5%3a19.03.8~3-0~raspbian-buster_armhf.deb ...\r\nUnpacking docker-ce (5:19.03.8~3-0~raspbian-buster) over (5:19.03.10~3-0~raspbian-buster) ...\r\nSetting up aufs-tools (1:4.14+20190211-1) ...\r\nSetting up dkms (2.6.1-4) ...\r\nSetting up docker-ce (5:19.03.8~3-0~raspbian-buster) ...\r\nSetting up aufs-dkms (4.19+20190211-1) ...\r\nLoading new aufs-4.19+20190211 DKMS files...\r\nIt is likely that 4.19.118-v7l+ belongs to a chroot's host\r\nBuilding for 4.19.118+, 4.19.118-v7+, 4.19.118-v7l+ and 4.19.118-v8+\r\nBuilding initial module for 4.19.118+\r\nError! Bad return status for module build on kernel: 4.19.118+ (armv7l)\r\nConsult /var/lib/dkms/aufs/4.19+20190211/build/make.log for more information.\r\ndpkg: error processing package aufs-dkms (--configure):\r\n installed aufs-dkms package post-installation script subprocess returned error exit status 10\r\nProcessing triggers for systemd (241-7~deb10u4+rpi1) ...\r\nProcessing triggers for man-db (2.8.5-2) ...\r\nProcessing triggers for libc-bin (2.28-10+rpi1) ...\r\nErrors were encountered while processing:\r\n aufs-dkms\r\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following additional packages will be installed:", "  aufs-dkms aufs-tools dkms", "Suggested packages:", "  aufs-dev python3-apport menu", "Recommended packages:", "  cgroupfs-mount | cgroup-lite libltdl7", "The following NEW packages will be installed:", "  aufs-dkms aufs-tools dkms", "The following packages will be DOWNGRADED:", "  docker-ce", "0 upgraded, 3 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.", "Need to get 0 B/15.3 MB of archives.", "After this operation, 5305 kB of additional disk space will be used.", "Selecting previously unselected package dkms.", "(Reading database ... ", "(Reading database ... 5%", "(Reading database ... 10%", "(Reading database ... 15%", "(Reading database ... 20%", "(Reading database ... 25%", "(Reading database ... 30%", "(Reading database ... 35%", "(Reading database ... 40%", "(Reading database ... 45%", "(Reading database ... 50%", "(Reading database ... 55%", "(Reading database ... 60%", "(Reading database ... 65%", "(Reading database ... 70%", "(Reading database ... 75%", "(Reading database ... 80%", "(Reading database ... 85%", "(Reading database ... 90%", "(Reading database ... 95%", "(Reading database ... 100%", "(Reading database ... 97112 files and directories currently installed.)", "Preparing to unpack .../archives/dkms_2.6.1-4_all.deb ...", "Unpacking dkms (2.6.1-4) ...", "Selecting previously unselected package aufs-dkms.", "Preparing to unpack .../aufs-dkms_4.19+20190211-1_all.deb ...", "Unpacking aufs-dkms (4.19+20190211-1) ...", "Selecting previously unselected package aufs-tools.", "Preparing to unpack .../aufs-tools_1%3a4.14+20190211-1_armhf.deb ...", "Unpacking aufs-tools (1:4.14+20190211-1) ...", "dpkg: warning: downgrading docker-ce from 5:19.03.10~3-0~raspbian-buster to 5:19.03.8~3-0~raspbian-buster", "Preparing to unpack .../docker-ce_5%3a19.03.8~3-0~raspbian-buster_armhf.deb ...", "Unpacking docker-ce (5:19.03.8~3-0~raspbian-buster) over (5:19.03.10~3-0~raspbian-buster) ...", "Setting up aufs-tools (1:4.14+20190211-1) ...", "Setting up dkms (2.6.1-4) ...", "Setting up docker-ce (5:19.03.8~3-0~raspbian-buster) ...", "Setting up aufs-dkms (4.19+20190211-1) ...", "Loading new aufs-4.19+20190211 DKMS files...", "It is likely that 4.19.118-v7l+ belongs to a chroot's host", "Building for 4.19.118+, 4.19.118-v7+, 4.19.118-v7l+ and 4.19.118-v8+", "Building initial module for 4.19.118+", "Error! Bad return status for module build on kernel: 4.19.118+ (armv7l)", "Consult /var/lib/dkms/aufs/4.19+20190211/build/make.log for more information.", "dpkg: error processing package aufs-dkms (--configure):", " installed aufs-dkms package post-installation script subprocess returned error exit status 10", "Processing triggers for systemd (241-7~deb10u4+rpi1) ...", "Processing triggers for man-db (2.8.5-2) ...", "Processing triggers for libc-bin (2.28-10+rpi1) ...", "Errors were encountered while processing:", " aufs-dkms"]}
```

I believe it is related to this [issue](https://github.com/raspberrypi/linux/issues/3021) with the aufs-dkms package.

geerlingguy, 2020-06-07:
@jtcmedia - I've been testing this on my Pi 4 running the latest image (2020-05-27 32-bit, and also the 64-bit beta that is downloadable from the forums), and I have not been able to reproduce the issue.

geerlingguy, 2020-06-07:
Well well well... I am incorrect. I was testing an older OS on the first run for 32-bits... and that version was succeeding for some reason.

I just re-tried again on a fresh image flash on Pi OS 32-bit and am hitting the same error:

```
TASK [geerlingguy.docker_arm : Make sure Docker CE is the version specified.] ******************************************
fatal: [10.0.100.52]: FAILED! => changed=false 
  cache_update_time: 1591497096
  cache_updated: false
  msg: |-
    '/usr/bin/apt-get -y -o "Dpkg::Options::=--force-confdef" -o "Dpkg::Options::=--force-confold"   --force-yes   install 'docker-ce=5:19.03.8~3-0~raspbian-buster'' failed: W: --force-yes is deprecated, use one of the options starting with --allow instead.
    E: Sub-process /usr/bin/dpkg returned an error code (1)
  rc: 100
  stderr: |-
    W: --force-yes is deprecated, use one of the options starting with --allow instead.
    E: Sub-process /usr/bin/dpkg returned an error code (1)
  stderr_lines: <omitted>
  stdout: |-
    Reading package lists...
    Building dependency tree...
    Reading state information...
    The following additional packages will be installed:
      aufs-dkms aufs-tools dkms raspberrypi-kernel-headers
    Suggested packages:
      aufs-dev python3-apport menu
    Recommended packages:
      cgroupfs-mount | cgroup-lite
    The following NEW packages will be installed:
      aufs-dkms aufs-tools dkms raspberrypi-kernel-headers
    The following packages will be DOWNGRADED:
      docker-ce
    0 upgraded, 4 newly installed, 1 downgraded, 0 to remove and 21 not upgraded.
    Need to get 40.2 MB of archives.
    After this operation, 168 MB of additional disk space will be used.
    Get:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel-headers armhf 1.20200601-1 [24.9 MB]
    Get:2 https://download.docker.com/linux/raspbian buster/stable armhf docker-ce armhf 5:19.03.8~3-0~raspbian-buster [15.0 MB]
    Get:3 http://mirrors.gigenet.com/raspbian/raspbian buster/main armhf dkms all 2.6.1-4 [74.4 kB]
    Get:4 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf aufs-dkms all 4.19+20190211-1 [183 kB]
    Get:5 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf aufs-tools armhf 1:4.14+20190211-1 [103 kB]
    Fetched 40.2 MB in 3s (11.5 MB/s)
    Selecting previously unselected package dkms.
    (Reading database ... (Reading database ... 5%(Reading database ... 10%(Reading database ... 15%(Reading database ... 20%(Reading database ... 25%(Reading database ... 30%(Reading database ... 35%(Reading database ... 40%(Reading database ... 45%(Reading database ... 50%(Reading database ... 55%(Reading database ... 60%(Reading database ... 65%(Reading database ... 70%(Reading database ... 75%(Reading database ... 80%(Reading database ... 85%(Reading database ... 90%(Reading database ... 95%(Reading database ... 100%(Reading database ... 94822 files and directories currently installed.)
    Preparing to unpack .../archives/dkms_2.6.1-4_all.deb ...
    Unpacking dkms (2.6.1-4) ...
    Selecting previously unselected package aufs-dkms.
    Preparing to unpack .../aufs-dkms_4.19+20190211-1_all.deb ...
    Unpacking aufs-dkms (4.19+20190211-1) ...
    Selecting previously unselected package aufs-tools.
    Preparing to unpack .../aufs-tools_1%3a4.14+20190211-1_armhf.deb ...
    Unpacking aufs-tools (1:4.14+20190211-1) ...
    dpkg: warning: downgrading docker-ce from 5:19.03.11~3-0~raspbian-buster to 5:19.03.8~3-0~raspbian-buster
    Preparing to unpack .../docker-ce_5%3a19.03.8~3-0~raspbian-buster_armhf.deb ...
    Unpacking docker-ce (5:19.03.8~3-0~raspbian-buster) over (5:19.03.11~3-0~raspbian-buster) ...
    Selecting previously unselected package raspberrypi-kernel-headers.
    Preparing to unpack .../raspberrypi-kernel-headers_1.20200601-1_armhf.deb ...
    Unpacking raspberrypi-kernel-headers (1.20200601-1) ...
    Setting up aufs-tools (1:4.14+20190211-1) ...
    Setting up dkms (2.6.1-4) ...
    Setting up raspberrypi-kernel-headers (1.20200601-1) ...
    run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.118+
    run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.118-v7+
    run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.118-v7l+
    run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.118-v8+
    Setting up docker-ce (5:19.03.8~3-0~raspbian-buster) ...
    Setting up aufs-dkms (4.19+20190211-1) ...
    Loading new aufs-4.19+20190211 DKMS files...
    It is likely that 4.19.118-v7l+ belongs to a chroot's host
    Building for 4.19.118+, 4.19.118-v7+, 4.19.118-v7l+ and 4.19.118-v8+
    Building initial module for 4.19.118+
    Error! Bad return status for module build on kernel: 4.19.118+ (armv7l)
    Consult /var/lib/dkms/aufs/4.19+20190211/build/make.log for more information.
    dpkg: error processing package aufs-dkms (--configure):
     installed aufs-dkms package post-installation script subprocess returned error exit status 10
    Processing triggers for systemd (241-7~deb10u4+rpi1) ...
    Processing triggers for man-db (2.8.5-2) ...
    Processing triggers for libc-bin (2.28-10+rpi1) ...
    Errors were encountered while processing:
     aufs-dkms
  stdout_lines: <omitted>
```

Sorry about doubting you for a second :D

geerlingguy, 2020-06-07:
Also related:

  - https://github.com/Azure/iotedge/issues/1642
  - https://github.com/docker/docker-install/issues/125

geerlingguy, 2020-06-07:
Released a `4.0.0` role version with the fix, since it's a potentially breaking change.

geerlingguy, 2020-06-07:
Also adjusting title, since this fix was not needed on the 64-bit version.

closed 2020-06-07 by geerlingguy

jtcmedia, 2020-06-08:
Glad you could reproduce it! Thanks for the role.

commit a07e6bee (2020-06-07): Merge pull request #12 from jtcmedia/fix-for-issue-11

add install_recommends: no, fixes #11
commit a31a10f7 (2020-06-07): add install_recommends: no, fixes #11