geerlingguy/ansible-role-docker_arm #14: Raspberry Pi OS - No package matching 'docker-ce' is available

jmedlin, 2020-07-28:
I flashed a fresh copy of Raspberry Pi OS (2020-05-27-raspios-buster-lite-armhf.img) to the micro SD card then booted it in a Raspberry Pi 3B+.  I enabled SSH and ran the following commands to prep it for Ansible:

```
pi@homeassistant-hub:~ $ cat .bash_history
sudo raspi-config
sudo apt-get update
sudo visudo
sudo apt-get upgrade
sudo reboot
```

My playbook is as follows:
```
---
- hosts: homeassistant_hub
  become: yes
  gather_facts: True
  vars:
    pip_package: python3-pip
    docker_install_compose: true
    docker_install_recommends: true
    docker_users:
      - pi

  roles:
    - geerlingguy.pip
    - geerlingguy.docker
```

This is the result of the playbook running:
```
➜  fort-edlin git:(master) ✗ ansible-galaxy list
# /Users/bi66pje/.ansible/roles
- geerlingguy.docker_arm, 4.0.0
- geerlingguy.docker, 2.8.1
- geerlingguy.nginx, 2.7.0
- geerlingguy.pip, 1.3.0
[WARNING]: - the configured path /usr/share/ansible/roles does not exist.
[WARNING]: - the configured path /etc/ansible/roles does not exist.
➜  fort-edlin git:(master) ✗ ansible-playbook homeassistant_hub.yaml
 __________________________
< PLAY [homeassistant_hub] >
 --------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

 ________________________
< TASK [Gathering Facts] >
 ------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

[WARNING]: Platform linux on host 192.168.10.24 is using the discovered Python interpreter at /usr/bin/python, but future installation of another
Python interpreter could change this. See https://docs.ansible.com/ansible/2.9/reference_appendices/interpreter_discovery.html for more information.
ok: [192.168.10.24]
 ___________________________________________________
< TASK [geerlingguy.pip : Ensure Pip is installed.] >
 ---------------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

[WARNING]: Updating cache and auto-installing missing dependency: python-apt
changed: [192.168.10.24]
 _________________________________________________________
/ TASK [geerlingguy.pip : Ensure pip_install_packages are \
\ installed.]                                             /
 ---------------------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

 ___________________________________________
< TASK [geerlingguy.docker : include_tasks] >
 -------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

skipping: [192.168.10.24]
 ___________________________________________
< TASK [geerlingguy.docker : include_tasks] >
 -------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

included: /Users/bi66pje/.ansible/roles/geerlingguy.docker/tasks/setup-Debian.yml for 192.168.10.24
 __________________________________________________________
/ TASK [geerlingguy.docker : Ensure old versions of Docker \
\ are not installed.]                                      /
 ----------------------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

ok: [192.168.10.24]
 ____________________________________________________
/ TASK [geerlingguy.docker : Ensure dependencies are \
\ installed.]                                        /
 ----------------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

changed: [192.168.10.24]
 _________________________________________________
< TASK [geerlingguy.docker : Add Docker apt key.] >
 -------------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

changed: [192.168.10.24]
 _____________________________________________________________
/ TASK [geerlingguy.docker : Ensure curl is present (on older \
\ systems without SNI).]                                      /
 -------------------------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

skipping: [192.168.10.24]
 ____________________________________________________________
/ TASK [geerlingguy.docker : Add Docker apt key (alternative \
\ for older systems without SNI).]                           /
 ------------------------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

skipping: [192.168.10.24]
 ____________________________________________________
< TASK [geerlingguy.docker : Add Docker repository.] >
 ----------------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

changed: [192.168.10.24]
 _____________________________________________
< TASK [geerlingguy.docker : Install Docker.] >
 ---------------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

fatal: [192.168.10.24]: FAILED! => {"changed": false, "msg": "No package matching 'docker-ce' is available"}
 ____________
< PLAY RECAP >
 ------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||

192.168.10.24              : ok=7    changed=4    unreachable=0    failed=1    skipped=4    rescued=0    ignored=0
```

GeorgeGedox, 2020-09-23:
I can confirm that this role is broken on newer version of Raspbian Lite, mostly due to the move to nftables instead of iptables

muhittink, 2020-12-12:
Same problem with ubuntu and arm64. Only `docker.io` is available

davidluque1, 2020-12-30:
Same issue here.

geerlingguy, 2021-04-28:
Note: This role is called `geerlingguy.docker_arm`. If you use my regular `docker` role, it will not work on Raspberry Pi. Since this role is architecturally very different than the main `docker` role it is separate.

You'll need to update your playbook to use this role and not the regular docker role.