geerlingguy/ansible-role-docker_arm #16: ubuntu-20.04.1-preinstalled-server-arm64+raspi supported?

jhalbrecht, 2020-11-14:
FWIW I'm a noob to ansible not to linux. I have successfully run some ad-hoc commands and playbooks against a vagrant virtualbox vm and my own rpi4 cluster.

I'm booting 64bit `ubuntu-20.04.1-preinstalled-server-arm64+raspi` on my Raspberry Pi 4 SBCs. I set the following defaults

```
docker_version_armv7: ubuntu-20.04.1-preinstalled-server-arm7+raspi
docker_version_arm64: ubuntu-20.04.1-preinstalled-server-arm64+raspi
docker_version: "{{ docker_version_armv7 if 'armv7' in ansible_architecture else docker_version_arm64 }}" 
```
resulting in 'not found' in the `TASK [geerlingguy.docker_arm : Make sure Docker CE is the version specified.]`

is the rpi 64bit ubuntu compatible with ansible-role-docker_arm? If so how might I properly configure / overide the defaults?

Also is the rpi 4 arm v8?

geerlingguy, 2020-11-14:
The Raspberry Pi 4 does have an armv8 processor. Note that Pi OS is by default 32-bit (to maintain compatibility all the way back to the original Pi), but there is a 64-bit beta that is also available now.

jhalbrecht, 2020-11-14:
> is the rpi 64bit ubuntu compatible with ansible-role-docker_arm? If so how might I properly configure / overide the defaults?

How very not ansible, but I sshed in to the client host to find docker running and I accomplished the 'hello world' apparently the `TASK [geerlingguy.docker_arm : Make sure Docker CE is the version specified.]` failed and the sudo. Jeff in twitter indicates I should look for `docker_users` var to set user ubuntu rather than pi.

[image]

hassiweb, 2021-01-31:
In my environment, this playbook installed the version of `5:20.10.2~3-0~ubuntu-focal` in Ubuntu 64bit on RPi 4B.
Also, `5:19.03.9~3-0~debian-buster`, which is defined in `defualts/main.yml`, cannot be found using `apt-cache policy docker-ce` because of the difference `ansible_distribution` (Raspberry Pi OS shows it as `Debian`, and Ubuntu shows `Ubuntu`).
This is why the error occurred because the docker version is not matched with `docker_version_arm64: 5:19.03.9~3-0~debian-buster`.

So, I guess this playbook was made for Raspberry Pi OS or other Debian-based OSes, not Ubuntu.

closed 2021-06-02 by stale[bot]