geerlingguy/ansible-role-docker_arm #18: Dryrun failing because of missing check_mode: no option in validation task

callmeconnor, 2021-01-06:
Hi, just "--check"runned a tiny installation playbook calling your and noticed an error right at the 1nd task preventing successfull execution.

Expected behaviour:
```
ansible-playbook docker_install.yml --check
 [WARNING]: provided hosts list is empty, only localhost is available [...]

PLAY [Docker playbook] **************************************************************************
TASK [Gathering Facts] **************************************************************************
ok: [localhost]

TASK [geerlingguy.docker_arm : Check if Docker is already present.] *****************************
skipping: [localhost]
[...]
```

Actual behaviour:
```
ansible-playbook docker_install.yml --check
[WARNING]: provided hosts list is empty, only localhost is available [...]
 
PLAY [Docker playbook] **************************************************************************
TASK [Gathering Facts] **************************************************************************
ok: [localhost]

TASK [geerlingguy.docker_arm : Download Docker install convenience script.] *********************
fatal: [localhost]: 
FAILED! => {"msg": "The conditional check 'docker_command_result.rc == 1' failed. The error was: error while evaluating conditional (docker_command_result.rc == 1): 'dict object' has no attribute 'rc'\n\nThe error appears to have been in '/home/pi/.ansible/roles/geerlingguy.docker_arm/tasks/main.yml': line 8, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: Download Docker install convenience script.\n  ^ here\n"}
        to retry, use: --limit @/home/pi/docker_install.retry

PLAY RECAP **************************************************************************************
localhost                  : ok=1    changed=0    unreachable=0    failed=1
```
I'm quite familiar with this behaviour, that  commands, in this case in the Task: "geerlingguy.docker_arm : Check if Docker is already present." is skipped in a check run. AFAIK the command "which docker" is still the most convenient way to go here, in case a simple [validation task](https://docs.ansible.com/ansible/latest/user_guide/playbooks_checkmode.html#using-check-mode) is done. 

So, as in many other occasions I simply added the option 
"check_mode: no"
to the task wich produces the expected dryrun behaviour which now works like a charm :) . Would you like to add that here also? Hope I get githubs PR workflow figured out quickly, will add this then ;)

geerlingguy, 2021-04-07:
I'd like to make sure the role works in check mode, so feel free to file a PR. Thanks!

callmeconnor, 2021-04-11:
OK, polishing it up now... will reply here once linting is OK. Sadly, i missed that your project lint's as it  should, but initially didn't care too much...

callmeconnor, 2021-04-11:
OK, git groundwork done. Noticed you removed linting again, so also not relevant for my pullrequest anymore, I guess...

Glorious one liner pull request 19  is still existing ;)  . Ready for you review and merge: https://github.com/geerlingguy/ansible-role-docker_arm/pull/19

callmeconnor, 2021-04-11:
hmm, not too deep into it, but: 

```
Molecule default > converge
WARNING  An error occurred during the test sequence action: 'converge'. Cleaning up.
```

callmeconnor, 2021-04-11:
Proof of work OK, did a checkrun, current output see below:

`ansible-playbook raspi-docker.yml --check --connection=local
 [WARNING]: Found both group and host with same name: localhost

Executing playbook raspi-docker.yml

- Base Config for pi -
Gathering Facts...
Sunday 11 April 2021  14:39:30 +0200 (0:00:00.451)       0:00:00.451 ********** 
  localhost ok
Check if Docker is already present....
Sunday 11 April 2021  14:39:36 +0200 (0:00:05.333)       0:00:05.785 ********** 
  localhost ok | stdout: /usr/bin/docker
Download Docker install convenience script....
Sunday 11 April 2021  14:39:37 +0200 (0:00:01.709)       0:00:07.494 ********** 
Run Docker install convenience script....
Sunday 11 April 2021  14:39:38 +0200 (0:00:00.262)       0:00:07.757 ********** 
Make sure Docker CE is the version specified....
Sunday 11 April 2021  14:39:38 +0200 (0:00:00.260)       0:00:08.018 ********** 
  localhost done | stdout: Reading package lists...
Building dependency tree...
Reading state information...
Recommended packages:
  cgroupfs-mount | cgroup-lite libltdl7
The following packages will be DOWNGRADED:
  docker-ce
0 upgraded, 0 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
Inst docker-ce [5:20.10.5~3-0~raspbian-buster] (5:19.03.12~3-0~raspbian-buster Docker CE:buster [armhf])
Conf docker-ce (5:19.03.12~3-0~raspbian-buster Docker CE:buster [armhf])
 | stderr: W: --force-yes is deprecated, use one of the options starting with --allow instead.

Ensure containerd service dir exists....
Sunday 11 April 2021  14:39:48 +0200 (0:00:09.624)       0:00:17.643 ********** 
Add shim to ensure Docker can start in all environments....
Sunday 11 April 2021  14:39:48 +0200 (0:00:00.305)       0:00:17.948 ********** 
Reload systemd daemon if template is changed....
Sunday 11 April 2021  14:39:48 +0200 (0:00:00.283)       0:00:18.232 ********** 
Ensure Docker is started....
Sunday 11 April 2021  14:39:48 +0200 (0:00:00.290)       0:00:18.522 ********** 
  localhost ok
Install Docker Compose using Pip....
Sunday 11 April 2021  14:39:52 +0200 (0:00:03.541)       0:00:22.064 ********** 
Ensure docker users are added to the docker group....
Sunday 11 April 2021  14:39:52 +0200 (0:00:00.260)       0:00:22.325 ********** 
  localhost ok

- Play recap -
  localhost                  : ok=5    changed=1    unreachable=0    failed=0   
Sunday 11 April 2021  14:39:54 +0200 (0:00:02.300)       0:00:24.625 ********** 
=============================================================================== 
ansible-role-docker_arm : Make sure Docker CE is the version specified. ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ 9.62s
Gathering Facts -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 5.33s
ansible-role-docker_arm : Ensure Docker is started. -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 3.54s
ansible-role-docker_arm : Ensure docker users are added to the docker group. ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 2.30s
ansible-role-docker_arm : Check if Docker is already present. ---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 1.71s
ansible-role-docker_arm : Ensure containerd service dir exists. -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 0.31s
ansible-role-docker_arm : Reload systemd daemon if template is changed. ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ 0.29s
ansible-role-docker_arm : Add shim to ensure Docker can start in all environments. ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 0.28s
ansible-role-docker_arm : Download Docker install convenience script. -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 0.26s
ansible-role-docker_arm : Install Docker Compose using Pip. ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ 0.26s
ansible-role-docker_arm : Run Docker install convenience script. ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 0.26s
`

geerlingguy, 2021-04-28:
Fixed via #19

closed 2021-04-28 by geerlingguy