geerlingguy/ansible-role-docker_arm #20: 'ERROR! no action detected in task.' when using 'docker_compose'

Freekers, 2021-01-19:
I'm trying to write a playbook that uses 'docker_compose' as an action in a task in my playbook, but it keeps telling me there is no action detected in the task. The same syntax in combination with your non ARM-docker role which works fine though. 
Is there a different in syntax on how to use 'docker_compose' as an action in a task for the ARM-docker role compared to the non-ARM docker role?

Steps to reproduce:

Suppose I have the following playbook:

```
- hosts: adguard
  become: true
  vars:
    pip_package: python3-pip
    docker_install_compose: true
    docker_users:
      - adguard
  roles:
    - geerlingguy.pip
    - geerlingguy.docker_arm

- hosts: adguard
  become: true
  tasks:
  - name: Copy Docker Compose
    copy:
      src: docker-compose.yml
      dest: /opt/adguard/docker-compose.yml

  - name: Deploy Stack using Docker Compose
    docker_compose:
      project_src: /opt/adguard
      state: present
```

Alongside the following requirements file:

```
- src: geerlingguy.docker_arm
- src: geerlingguy.pip
- src: oefenweb.rc_local
```

Running `ansible-galaxy install -r requirements.yml --force` outputs:
```
- changing role geerlingguy.docker_arm from 4.0.0 to unspecified
- downloading role 'docker_arm', owned by geerlingguy
- downloading role from https://github.com/geerlingguy/ansible-role-docker_arm/archive/4.0.0.tar.gz
- extracting geerlingguy.docker_arm to /home/pirate/.ansible/roles/geerlingguy.docker_arm
- geerlingguy.docker_arm (4.0.0) was installed successfully
- downloading role 'pip', owned by geerlingguy
- downloading role from https://github.com/geerlingguy/ansible-role-pip/archive/2.0.0.tar.gz
- extracting geerlingguy.pip to /home/pirate/.ansible/roles/geerlingguy.pip
- geerlingguy.pip (2.0.0) was installed successfully
- changing role oefenweb.rc_local from v1.0.40 to unspecified
- downloading role 'rc_local', owned by oefenweb
- downloading role from https://github.com/Oefenweb/ansible-rc-local/archive/v1.0.40.tar.gz
- extracting oefenweb.rc_local to /home/pirate/.ansible/roles/oefenweb.rc_local
- oefenweb.rc_local (v1.0.40) was installed successfully
```

Afterwards, the playbook is run as follows: ``ansible-playbook playbook.yml --ask-become-pass -vvv``
which gives the following output:
```
ansible-playbook 2.7.7
  config file = /home/pirate/ansible-adguard/ansible.cfg
  configured module search path = ['/home/pirate/.ansible/plugins/modules', '/usr/share/ansible/plugins/modules']
  ansible python module location = /usr/lib/python3/dist-packages/ansible
  executable location = /usr/bin/ansible-playbook
  python version = 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]
Using /home/pirate/ansible-adguard/ansible.cfg as config file
SUDO password:
/home/pirate/ansible-adguard/hosts did not meet host_list requirements, check plugin documentation if this is unexpected
/home/pirate/ansible-adguard/hosts did not meet script requirements, check plugin documentation if this is unexpected
Parsed /home/pirate/ansible-adguard/hosts inventory source with ini plugin
statically imported: /home/pirate/.ansible/roles/geerlingguy.docker_arm/tasks/docker-1809-shim.yml
ERROR! no action detected in task. This often indicates a misspelled module name, or incorrect module path.

The error appears to have been in '/home/pirate/ansible-adguard/playbook.yml': line 215, column 5, but may
be elsewhere in the file depending on the exact syntax problem.

The offending line appears to be:

  - name: Deploy Stack using Docker Compose
    ^ here
```

Additional debug information from the user that reported this to me:
Raspbian GNU/Linux 10 (buster)
Docker version 20.10.2, build 2291f61
docker-compose version 1.26.1, build 634eb50

Thank you!

tgallacher, 2021-02-03:
I think you might be missing the Docker python SDK:

```yml
- hosts: adguard
  become: true
  vars:
    pip_package: python3-pip
    pip_install_packages:                                         <---- Add this
        - name: docker                                            <---- and this
    docker_install_compose: true
    docker_users:
      - adguard
  roles:
    - geerlingguy.pip
    - geerlingguy.docker_arm

- hosts: adguard
  become: true
  tasks:
  ....
```

This is then installed via the `geerlingguy.pip` role.

Freekers, 2021-02-05:
Thanks for replying, @tgallacher 
I added both lines, but sadly the issue remains

```
ansible-playbook playbook.yml --ask-become-pass -vvv
ansible-playbook 2.7.7
  config file = /home/pi/ansible-adguard/ansible.cfg
  configured module search path = ['/home/pi/.ansible/plugins/modules', '/usr/share/ansible/plugins/modules']
  ansible python module location = /usr/lib/python3/dist-packages/ansible
  executable location = /usr/bin/ansible-playbook
  python version = 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]
Using /home/pi/ansible-adguard/ansible.cfg as config file
SUDO password:
/home/pi/ansible-adguard/hosts did not meet host_list requirements, check plugin documentation if this is unexpected
/home/pi/ansible-adguard/hosts did not meet script requirements, check plugin documentation if this is unexpected
Parsed /home/pi/ansible-adguard/hosts inventory source with ini plugin
statically imported: /home/pi/.ansible/roles/geerlingguy.docker_arm/tasks/docker-1809-shim.yml
ERROR! no action detected in task. This often indicates a misspelled module name, or incorrect module path.

The error appears to have been in '/home/pi/ansible-adguard/playbook.yml': line 217, column 5, but may
be elsewhere in the file depending on the exact syntax problem.

The offending line appears to be:

  - name: Deploy Stack using Docker Compose
    ^ here
```

tgallacher, 2021-02-06:
Ah. I think you might need to use `docker_service` instead of `docker_compose`; It was renamed in Ansible v2.8, but you appear to be on v2.7.

Freekers, 2021-02-22:
@tgallacher D'oh, you were right. Thank you!

closed 2021-02-22 by Freekers