geerlingguy/ansible-role-docker_arm #23: Error:  /var/run/docker.sock: connect: permission denied

typekpb, 2021-05-24:
After successfull role application on my rpi:
```
===============================================================================
geerlingguy.docker_arm : Run Docker install convenience script. ---------------------------------------------------------------------------------------------------------------------------------------------------------------------- 61.03s
geerlingguy.docker_arm : Install Docker Compose using Pip. --------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 27.54s
geerlingguy.docker_arm : Make sure Docker CE is the version specified. --------------------------------------------------------------------------------------------------------------------------------------------------------------- 17.59s
Gathering Facts ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 2.26s
geerlingguy.docker_arm : Ensure Docker is started. ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ 1.62s
geerlingguy.docker_arm : Download Docker install convenience script. ------------------------------------------------------------------------------------------------------------------------------------------------------------------ 1.20s
geerlingguy.docker_arm : Check if Docker is already present. -------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 0.69s
geerlingguy.docker_arm : Ensure docker users are added to the docker group. ----------------------------------------------------------------------------------------------------------------------------------------------------------- 0.11s
geerlingguy.docker_arm : Ensure containerd service dir exists. ------------------------------------------------------------------------------------------------------------------------------------------------------------------------ 0.05s
geerlingguy.docker_arm : Reload systemd daemon if template is changed. ---------------------------------------------------------------------------------------------------------------------------------------------------------------- 0.05s
geerlingguy.docker_arm : Add shim to ensure Docker can start in all environments. ----------------------------------------------------------------------------------------------------------------------------------------------------- 0.05s
```

Once trying to run docker after ansible role passed on target machine, I get:
```
docker ps
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get http://%2Fvar%2Frun%2Fdocker.sock/v1.24/containers/json: dial unix /var/run/docker.sock: connect: permission denied
```

geerlingguy, 2021-05-24:
@typekpb - Two questions: first, did you add a user to the docker_users list so this role can add your user account to it? Second, did you log out and log back in so the new group permissions are present on your account?

typekpb, 2021-05-24:
setting `docker_users` resolved my issue, thanks!

closed 2021-05-24 by typekpb