geerlingguy/ansible-role-docker_arm #25: feature wish: use the latest docker version for RaspianOS 32 bit

cybermcm, 2021-08-02:
Currently, the docker version for RaspianOS (32 bit) is 20.10.7, but this role installs 19.03.12.
Maybe it is possible to extract the current version number from get.docker.com? Sorry I'm no programmer, so I've no idea how to do that....

geerlingguy, 2021-08-04:
It's a matter of updating defaults to the latest versions (see https://github.com/geerlingguy/ansible-role-docker_arm#role-variables) — that could be done fairly easily in a PR, but someone needs to get the exact values to update them all.

cybermcm, 2021-08-08:
OK understood, but I never did that before :-(
I don't know how to create a PR and even before I tried to get the correct version numbers, but I failed (didn't find the right place to look for)... sorry, total noob here....

geerlingguy, 2021-10-29:
Just pushed an update for the latest version on Debian 10 64-bit Pi OS.

@cybermcm - Can you run `apt-cache madison docker-ce` on your Pi to see what version strings are available? Also `uname -a` to get the platform architecture.

cybermcm, 2021-10-30:
That I can do, thanks for taking time to look into it:
```apt-cache madison docker-ce
 docker-ce | 5:20.10.10~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.9~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.8~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.7~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.6~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.5~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.4~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.3~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.2~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.1~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:20.10.0~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.15~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.14~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.13~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.12~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.11~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.10~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.9~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.8~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.7~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.6~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.5~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.4~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.3~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.2~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
 docker-ce | 5:19.03.1~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
```
```
uname -a
Linux mail 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
```

fernandopasik, 2021-11-29:
I've been getting the latest version and storing it in a var and then running jeff's role

```
    - name: Get latest Docker CE version
      shell: apt list -a docker-ce | head -n 2 | tail -1 | awk '{print $2}'
      register: docker_ce_latest
    - set_fact:
        docker_ce_latest: '{{docker_ce_latest.stdout}}'
    - import_role:
        name: geerlingguy.docker_arm
      vars:
        docker_version_armv6: '{{docker_ce_latest}}'
        docker_version_armv7: '{{docker_ce_latest}}'
        docker_version_arm64: '{{docker_ce_latest}}'
```

fernandopasik, 2021-11-29:
I've realised this can not be done without running first the convenience script install  or manually setting up the docker repo.

@geerlingguy I think the docker install convenience script installs already the latest version, would it be possible to make optional the step 

```
- name: Make sure Docker CE is the version specified.
  apt:
    name: "docker-ce={{ docker_version }}"
    state: present
    force: true
    install_recommends: "{{ docker_install_recommends }}"
```

geerlingguy, 2021-11-29:
@fernandopasik - I'd be okay with that, maybe set `docker_version` default to `latest`, and in this task, use a `when` to say `{{ docker_version is not 'latest' }}` (something like that).

fernandopasik, 2021-11-29:
Pushed a PR to avoid step when `docker_version` is set to `latest`

PR #33 (2021-11-29): feat: avoid ensure version step when docker_version is latest
fixes #25
closed 2021-11-29 by geerlingguy