geerlingguy/ansible-role-docker_arm #27: not able to install docker-compose on rpi 4 64bit

stsch83, 2021-08-12:
I got this errors when i was trying to install docker-compose with this script.

pip_package: python3-pip
docker_pip_executable: pip3

pre_tasks:
    - apt: update_cache=true

  roles:
    - geerlingguy.pip
    - geerlingguy.docker_arm

TASK [geerlingguy.docker_arm : Install Docker Compose using Pip.] ****************************************************************************************************************************************
fatal: [192.168.65.141]: FAILED! => {"changed": false, "cmd": ["/usr/bin/pip3", "install", "docker-compose"], "msg": "stdout: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\nCollecting docker-compose\n  Using cached https://files.pythonhosted.org/packages/f3/3e/ca05e486d44e38eb495ca60b8ca526b192071717387346ed1031ecf78966/docker_compose-1.29.2-py2.py3-none-any.whl\nCollecting PyYAML<6,>=3.10 (from docker-compose)\n  Using cached https://files.pythonhosted.org/packages/a0/a4/d63f2d7597e1a4b55aa3b4d6c5b029991d3b824b5bd331af8d4ab1ed687d/PyYAML-5.4.1.tar.gz\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'done'\nCollecting distro<2,>=1.5.0 (from docker-compose)\n  Using cached https://files.pythonhosted.org/packages/b3/8d/a0a5c389d76f90c766e956515d34c3408a1e18f60fbaa08221d1f6b87490/distro-1.6.0-py2.py3-none-any.whl\nCollecting jsonschema<4,>=2.5.1 (from docker-compose)\n  Using cached https://files.pythonhosted.org/packages/c5/8f/51e89ce52a085483359217bc72cdbf6e75ee595d5b1d4b5ade40c7e018b8/jsonschema-3.2.0-py2.py3-none-any.whl\nCollecting docopt<1,>=0.6.1 (from docker-compose)\n  Using cached https://www.piwheels.org/simple/docopt/docopt-0.6.2-py2.py3-none-any.whl\nCollecting python-dotenv<1,>=0.13.0 (from docker-compose)\n  Using cached https://files.pythonhosted.org/packages/f5/d6/4b6268fad900fcb064e4344aa563b22688f0b38dcd857b500b2b5cc445c6/python_dotenv-0.19.0-py2.py3-none-any.whl\nRequirement already satisfied: requests<3,>=2.20.0 in /usr/lib/python3/dist-packages (from docker-compose) (2.21.0)\nCollecting texttable<2,>=0.9.0 (from docker-compose)\n  Using cached https://files.pythonhosted.org/packages/75/23/8170868d04b153b1b9ed1bb84348212bb4a08f31b292ef9d7f6ea648fd49/texttable-1.6.4-py2.py3-none-any.whl\nCollecting cached-property<2,>=1.2.0; python_version < \"3.8\" (from docker-compose)\n  Using cached https://files.pythonhosted.org/packages/48/19/f2090f7dad41e225c7f2326e4cfe6fff49e57dedb5b53636c9551f86b069/cached_property-1.5.2-py2.py3-none-any.whl\nCollecting docker[ssh]>=5 (from docker-compose)\n  Using cached https://files.pythonhosted.org/packages/b2/5a/f988909dfed18c1ac42ad8d9e611e6c5657e270aa6eb68559985dbb69c13/docker-5.0.0-py2.py3-none-any.whl\nCollecting websocket-client<1,>=0.32.0 (from docker-compose)\n  Using cached https://files.pythonhosted.org/packages/f7/0c/d52a2a63512a613817846d430d16a8fbe5ea56dd889e89c68facf6b91cb6/websocket_client-0.59.0-py2.py3-none-any.whl\nCollecting dockerpty<1,>=0.4.1 (from docker-compose)\n  Using cached https://www.piwheels.org/simple/dockerpty/dockerpty-0.4.1-py3-none-any.whl\nCollecting pyrsistent>=0.14.0 (from jsonschema<4,>=2.5.1->docker-compose)\n  Using cached https://files.pythonhosted.org/packages/f4/d7/0fa558c4fb00f15aabc6d42d365fcca7a15fcc1091cd0f5784a14f390b7f/pyrsistent-0.18.0.tar.gz\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'done'\nCollecting importlib-metadata; python_version < \"3.8\" (from jsonschema<4,>=2.5.1->docker-compose)\n  Using cached https://files.pythonhosted.org/packages/07/76/c4674c460f5ff4b5f7a962214e46295e20504dfde9fcba78fd728dfe2ac9/importlib_metadata-4.6.3-py3-none-any.whl\nCollecting attrs>=17.4.0 (from jsonschema<4,>=2.5.1->docker-compose)\n  Using cached https://files.pythonhosted.org/packages/20/a9/ba6f1cd1a1517ff022b35acd6a7e4246371dfab08b8e42b829b6d07913cc/attrs-21.2.0-py2.py3-none-any.whl\nRequirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from jsonschema<4,>=2.5.1->docker-compose) (40.8.0)\nRequirement already satisfied: six>=1.11.0 in /usr/lib/python3/dist-packages (from jsonschema<4,>=2.5.1->docker-compose) (1.12.0)\nCollecting paramiko>=2.4.2; extra == \"ssh\" (from docker[ssh]>=5->docker-compose)\n  Using cached https://files.pythonhosted.org/packages/95/19/124e9287b43e6ff3ebb9cdea3e5e8e88475a873c05ccdf8b7e20d2c4201e/paramiko-2.7.2-py2.py3-none-any.whl\nCollecting typing-extensions>=3.6.4; python_version < \"3.8\" (from importlib-metadata; python_version < \"3.8\"->jsonschema<4,>=2.5.1->docker-compose)\n  Using cached https://files.pythonhosted.org/packages/2e/35/6c4fff5ab443b57116cb1aad46421fb719bed2825664e8fe77d66d99bcbc/typing_extensions-3.10.0.0-py3-none-any.whl\nCollecting zipp>=0.5 (from importlib-metadata; python_version < \"3.8\"->jsonschema<4,>=2.5.1->docker-compose)\n  Using cached https://files.pythonhosted.org/packages/92/d9/89f433969fb8dc5b9cbdd4b4deb587720ec1aeb59a020cf15002b9593eef/zipp-3.5.0-py3-none-any.whl\nCollecting pynacl>=1.0.1 (from paramiko>=2.4.2; extra == \"ssh\"->docker[ssh]>=5->docker-compose)\n  Downloading https://files.pythonhosted.org/packages/cf/5a/25aeb636baeceab15c8e57e66b8aa930c011ec1c035f284170cacb05025e/PyNaCl-1.4.0.tar.gz (3.4MB)\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'error'\n  Complete output from command /usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-zo0h_3b2 --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools>=40.8.0 wheel \"cffi>=1.4.1; python_implementation != 'PyPy'\":\n  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple\n  Collecting setuptools>=40.8.0\n    Using cached https://files.pythonhosted.org/packages/bd/25/5bdf7f1adeebd4e3fa76b2e2f045ae53ee208e40a4231ad0f0c3007e4353/setuptools-57.4.0-py3-none-any.whl\n  Collecting wheel\n    Using cached https://files.pythonhosted.org/packages/04/80/cad93b40262f5d09f6de82adbee452fd43cdff60830b56a74c5930f7e277/wheel-0.37.0-py2.py3-none-any.whl\n  Collecting cffi>=1.4.1\n    Using cached https://files.pythonhosted.org/packages/2e/92/87bb61538d7e60da8a7ec247dc048f7671afe17016cd0008b3b710012804/cffi-1.14.6.tar.gz\n  Collecting pycparser (from cffi>=1.4.1)\n    Using cached https://files.pythonhosted.org/packages/ae/e7/d9c3a176ca4b02024debf82342dab36efadfc5776f9c8db077e8f6e71821/pycparser-2.20-py2.py3-none-any.whl\n  Building wheels for collected packages: cffi\n    Running setup.py bdist_wheel for cffi: started\n    Running setup.py bdist_wheel for cffi: finished with status 'error'\n    Complete output from command /usr/bin/python3 -u -c \"import setuptools, tokenize;__file__='/tmp/pip-install-s43ezo45/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\\r\\n', '\\n');f.close();exec(compile(code, __file__, 'exec'))\" bdist_wheel -d /tmp/pip-wheel-ci3bbmz3 --python-tag cp37:\n    Package libffi was not found in the pkg-config search path.\n    Perhaps you should add the directory containing `libffi.pc'\n    to the PKG_CONFIG_PATH environment variable\n    No package 'libffi' found\n    Package libffi was not found in the pkg-config search path.\n    Perhaps you should add the directory containing `libffi.pc'\n    to the PKG_CONFIG_PATH environment variable\n    No package 'libffi' found\n    Package libffi was not found in the pkg-config search path.\n    Perhaps you should add the directory containing `libffi.pc'\n    to the PKG_CONFIG_PATH environment variable\n    No package 'libffi' found\n    Package libffi was not found in the pkg-config search path.\n    Perhaps you should add the directory containing `libffi.pc'\n    to the PKG_CONFIG_PATH environment variable\n    No package 'libffi' found\n    Package libffi was not found in the pkg-config search path.\n    Perhaps you should add the directory containing `libffi.pc'\n    to the PKG_CONFIG_PATH environment variable\n    No package 'libffi' found\n    running bdist_wheel\n    running build\n    running build_py\n    creating build\n    creating build/lib.linux-aarch64-3.7\n    creating build/lib.linux-aarch64-3.7/cffi\n    copying cffi/ffiplatform.py -> build/lib.linux-aarch64-3.7/cffi\n    copying cffi/commontypes.py -> build/lib.linux-aarch64-3.7/cffi\n    copying cffi/model.py -> build/lib.linux-aarch64-3.7/cffi\n    copying cff [... cut ...] -no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools>=40.8.0 wheel \"cffi>=1.4.1; python_implementation != 'PyPy'\"\" failed with error code 1 in None\n"}

google helps and i found this:

sudo apt-get install -y libffi-dev libssl-dev
sudo apt-get install -y python3 python3-pip
sudo apt-get remove python-configparser
sudo pip3 -v install docker-compose

and it works.

is it possible to include this in the current srcipt?

geerlingguy, 2021-08-12:
Are you also running the `geerlingguy.pip` role before this role as mentioned in the README? That I believe takes care of some of these issues at least.

stsch83, 2021-08-12:
yes i did.

geerlingguy, 2021-08-12:
Hmm... I'll have to re-test on a fresh Pi 4 on 64-bit Pi OS and confirm that this happens there. Last time I set up a fresh Pi 4 about 1 month ago I didn't run into any problems using just this role and the pip role to get Docker set up.

stsch83, 2021-08-12:
> Hmm... I'll have to re-test on a fresh Pi 4 on 64-bit Pi OS and confirm that this happens there. Last time I set up a fresh Pi 4 about 1 month ago I didn't run into any problems using just this role and the pip role to get Docker set up.

Thx.

stsch83, 2021-08-17:
> Hmm... I'll have to re-test on a fresh Pi 4 on 64-bit Pi OS and confirm that this happens there. Last time I set up a fresh Pi 4 about 1 month ago I didn't run into any problems using just this role and the pip role to get Docker set up.

Did you find something, what is going wrong?

geerlingguy, 2021-10-29:
Testing it currently.

geerlingguy, 2021-10-29:
I just ran the role on a brand new Pi flashed with 64-bit Pi OS (fresh from Pi's site).

First I ran `apt-get update` and `apt-get upgrade -y`, then rebooted.

Then I ran this role, and the docker-compose pip install worked fine after a couple minutes' pause:

```
TASK [geerlingguy.docker_arm : Install Docker Compose using Pip.] ******************************************************
changed: [10.0.100.10]
```

I'm going to close the issue since i can't replicate it, but if you could provide a set of steps to reproduce the issue on a fresh Pi OS 64-bit install, I'll re-test with that set of instructions.

closed 2021-10-29 by geerlingguy