geerlingguy/ansible-role-docker_arm #29: problem with geerlingguy.docker_arm

quegzacov, 2021-09-01:
Hello, I use a raspberry pi 4 with a ubuntu server distro.
I'm not able to use geerlingguy.docker_arm's role in a playbook. I installed it with the command: `ansible-galaxy install geerlingguy.docker_arm`. After that, I write a playbook (docker.yml)
`---
- name: "Docker playbook"
  hosts: localhost
  connection: local
  become: yes
  vars:
    docker_install_compose: false
    docker_users:
      - ubuntu
  roles:
    - /home/ubuntu/.ansible/roles/geerlingguy.docker_arm
`
Finally, I ran the command  `ansible-playbook docker.yml` and that's the result

`[WARNING]: provided hosts list is empty, only localhost is available. Note that
the implicit localhost does not match 'all'

PLAY [Docker playbook] *********************************************************

TASK [Gathering Facts] *********************************************************
ok: [localhost]

TASK [/home/ubuntu/.ansible/roles/geerlingguy.docker_arm : Check if Docker is already present.] ***
ok: [localhost]

TASK [/home/ubuntu/.ansible/roles/geerlingguy.docker_arm : Download Docker install convenience script.] ***
skipping: [localhost]

TASK [/home/ubuntu/.ansible/roles/geerlingguy.docker_arm : Run Docker install convenience script.] ***
skipping: [localhost]

TASK [/home/ubuntu/.ansible/roles/geerlingguy.docker_arm : Make sure Docker CE is the version specified.] ***
fatal: [localhost]: FAILED! => {"cache_update_time": 1630497576, "cache_updated": false, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"   --force-yes   install 'docker-ce=5:19.03.9~3-0~debian-buster' -o APT::Install-Recommends=no' failed: W: --force-yes is deprecated, use one of the options starting with --allow instead.\nE: Version '5:19.03.9~3-0~debian-buster' for 'docker-ce' was not found\n", "rc": 100, "stderr": "W: --force-yes is deprecated, use one of the options starting with --allow instead.\nE: Version '5:19.03.9~3-0~debian-buster' for 'docker-ce' was not found\n", "stderr_lines": ["W: --force-yes is deprecated, use one of the options starting with --allow instead.", "E: Version '5:19.03.9~3-0~debian-buster' for 'docker-ce' was not found"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information..."]}

PLAY RECAP *********************************************************************
localhost                  : ok=2    changed=0    unreachable=0    failed=1    skipped=2    rescued=0    ignored=0   
`
Thanks in advance

pmaslana, 2021-09-22:
I ran into this issue earlier today. The version of the file that you specified is not listed in your apt cache. Therefore, run this command ```apt-cache madison docker-ce``` to see what is available. Choose one of those versions as set that as the ```docker_version``` variable.

geerlingguy, 2021-10-29:
Additionally... if it's skipping the convenience script, then it could be Docker was already installed on your system, and that's the issue. But more likely it's what @pmaslana is saying above.

I only 'officially' support this role on Raspberry Pi OS—some things may require changes on other OSes like Ubuntu.

closed 2021-10-29 by geerlingguy