geerlingguy/ansible-role-docker_arm #30: Version check fails for bullseye as defaults are set to buster

beardedfool, 2021-11-08:
In main.yml
- name: Make sure Docker CE is the version specified.

But the defaults are set to Buster
This works:
docker_version_armv6: "5:20.10.0~3-0~raspbian-{{ansible_distribution_release}}"
docker_version_armv7: "5:20.10.10~3-0~raspbian-{{ansible_distribution_release}}"
docker_version_arm64: "5:20.10.10~3-0~debian-{{ansible_distribution_release}}"

Sorry, don't know how to do a PR and wanted you opinion on whether this was a good way to solve it/ I may be misunderstanding the problem

Also, is there some way to just install the latest version, rather than specifying the exact version?

geerlingguy, 2021-11-08:
That might be an okay solution—though have you verified those packages are available on armv6 and armv7 builds of Pi OS? Usually the version numbers can be different from the ones for arm64.

beardedfool, 2021-11-09:
Took a while to set them up....

Seem to be in line...

Pi 3-arm7 bullyseye
docker-ce | 5:20.10.10~3-0~debian-bullseye | https://download.docker.com/linux/debian bullseye/stable arm64 Packages
Pi 4-aarch64 buster
docker-ce | 5:20.10.10~3-0~raspbian-bullseye | https://download.docker.com/linux/raspbian bullseye/stable armhf Packages

pi zeros armv6l
buster: 
docker-ce | 5:20.10.10~3-0~raspbian-buster | https://download.docker.com/linux/raspbian buster/stable armhf Packages
bullseye:
docker-ce | 5:20.10.10~3-0~raspbian-bullseye | https://download.docker.com/linux/raspbian bullseye/stable armhf Packages

Can you remember why you were doing that check, rather just in installing the latest?

And, whilst I have you there, thanks for all I've learned from you!

geerlingguy, 2021-11-09:
I know I was having trouble with many of the versions for ARM around the 19.03 era, so it was important to install the correct version at the time, otherwise you'd run into weird bugs with many containers.

I'm sure things have improved a bit, but for now, for stability's sake, I still like specifying the version so something drastic doesn't change from underneath my containers.

beardedfool, 2021-11-09:
Makes sense. 

One quick aside before I leave you in peace please just in case you have something using it...

In the 64 bullseye (only) the ansible_lsb.id is Debian

arm6/7 still show Raspbian

geerlingguy, 2021-11-09:
Yeah, that's a funky thing that has to do with the Pi OS being rebased for 64-bit only on top of Debian directly, instead of on the Raspbian fork.

smougenot, 2021-11-13:
Thank you for the fix @beardedfool 
For those ... like me ... that have copy/paste ruined by GH formating here are ... the code to copy
```
docker_version_armv6: "5:20.10.0~3-0~raspbian-{{ansible_distribution_release}}"
docker_version_armv7: "5:20.10.10~3-0~raspbian-{{ansible_distribution_release}}"
docker_version_arm64: "5:20.10.10~3-0~debian-{{ansible_distribution_release}}"
```

rizlas, 2021-11-14:
Thank you @beardedfool and thanks @smougenot for formatting it.

geerlingguy, 2021-11-29:
In this commit (https://github.com/geerlingguy/ansible-role-docker_arm/commit/c3c083b0d90677cbe24b51b7d09d8cffca046bd0), I set the default to `latest`, so now it will choose the proper version on whatever platform you are installing on (hopefully), and you can just override with a specific version if you need to.

closed 2021-11-29 by geerlingguy